Velorexe/Unity-Android-Bluetooth-Low-Energy #5: Empty Example Scene

NicoPowers, 2021-11-28:
Hello,

I want to use this package as it seems very cool and useful! But when I build to android and run there is no interface to scan for devices and connect and read/write. I only see an empty scene. I noticed there is a BLE manager, but its not obvious how I am supposed to get started connecting to a device.

Do I need to create a BLE Object and attach that script to a game object? Then what do I need to change for this to work? 

It would be nice if the example scene had a fake sensor or something to represent a device we would try to connect to, and then I can just change the UUID's for my device.

Thanks!

Velorexe, 2021-11-28:
Hi! 
Thanks for opening an issue about this. I'll get started on an example scene right away, since I haven't made it very clear with a tutorial how this package works. I'll notify you through this issue as soon as I got a demo!

Velorexe, 2021-11-30:
@NicoPowers I've updated the `development` branch in the repository with an example scene. I've tested this locally, but I'd like it if you could also test it out! In the coming days I'll add documentation to the example scene to make the flow more clearer.

closed 2022-01-24 by Velorexe