Velorexe/Unity-Android-Bluetooth-Low-Energy #14: The uuid becomes invalid after the reconnection

cmmmm660, 2022-11-21:
Hello @Velorexe 
I used an Android device to connect the ble Bluetooth module. After connecting the module, I would subscribe to the UUID of the module for sending and receiving data. Then I disconnected the module so that it would trigger the OnDisconnected event. The next problem is that the UUID that receives the data is subscribed but cannot receive the data sent by the module, and the same is true for the sending

Velorexe, 2022-11-22:
Hi,

I'm not sure if I'm following you here. You
- successfully connected to the BLE module
- subscribed to a characteristic
- disconnected
- reconnected to the BLE module
- subscribed to a characteristic

When reconnected a second time to the same characteristic, you don't receive nor can send any data?

cmmmm660, 2022-11-22:
Hi,
Yes! Before that, I still have a question. Why can’t I scan the scanned device name and address again?

cmmmm660, 2022-11-22:
The detailed process of reconnection is as follows
• 1. It is normal to connect the Bluetooth module at the beginning, and then subscribe to the UUID.
• 2. Next, I will unplug the module, which means the bluetooth signal will be disconnected. At this time, the OnDisconnected method will be triggered.

[image]

[image]

Velorexe, 2022-11-22:
Then I believe it's something in the Java library, when `subscribeFromGattCharacteristic` is called the flags of the characteristic are set to enable notifications, so it might be that  `unsubscribeFromGattCharacteristic` sets some flags on the characteristic or descriptor that prevents the device from sending new information.

I'll have a look at it.

As per your other question: I don't clear the list of already discovered devices from the Java library, but didn't add a method to retrieve the already discovered devices. I'll probably make a boolean on the scan command to clear the already discovered devices when scanning.

cmmmm660, 2022-11-23:
> 
expect,
thank you very much！！

cmmmm660, 2022-11-23:
hi,
I found that the jsonMessage disconnection or reconnection of the OnBleMessage method in the BleAdapter class will print out this information. Maybe you have judged this wrong in java. I hope it works for you. The bluetooth projects you create are very important to me. Hope to solve Thank you!

[image]

Velorexe, 2022-11-26:
I'm glad to hear that my BLE project is useful to you!
I'll take a look at the issue asap. Hope to get back to you soon.

Velorexe, 2022-11-27:
Hi @cmmmm660 

I've created a branch with a fix that worked when I reproduced your issue locally.
It seems to have been an issue with the order in which I disconnect the GATT server from the Java library. The `disconnect()` was called before the `close()`, so the server never really closed it's communication with the client (my BLE plugin).

Now it properly disconnects from the GATT server. I hope this will solve your problem.
The branch is called [`fix/subscribe-reconnect`](https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy/tree/fix/subscribe-reconnect)

cmmmm660, 2022-11-28:
Hi @Velorexe 
Thank you very much for your timely feedback and fixing the problems I encountered!
I use the latest branch, unfortunately there is still a problem with reconnecting, the error message is the same as before, but I also found that disconnecting the module after reconnecting does not seem to print the disconnection message, hope it is useful to you.

[image]
[image]

Velorexe, 2022-11-29:
Hi @cmmmm660 

Turns out, I added the changes to the branch with cherry picking, which only filters on lines and didn't track that I changed the Android library itself. I've added it and pushed it to the previously mentioned `fix/subscribe-reconnect` branch. Hope it works for you now!

cmmmm660, 2022-11-30:
Holle @Velorexe 
I used the fix/subscribe-reconnect branch and it reconnects now works fine, my problem was fixed, it was a big help, thank you so much for fixing it in time, it's too much for me Important, thank you very much! ! !

Velorexe, 2022-11-30:
No problem! Glad it fixed your problem and thanks for using my library.

closed 2022-11-30 by Velorexe