Velorexe/Unity-Android-Bluetooth-Low-Energy #16: gatt Descriptor is null

WimFaasen, 2022-12-06:
Hello @Velorexe,

I am trying to use your library on the quest. so far I have some progress with:
-scanning for devices (success)
-connecting to a device (and its services and characteristics) (success)

but during the subscribe to a characteristic I am now running into the issue that the service UUID and characteristic UUID is now accepted, but on line 384 in UnityAndroidBLE.java it crashes with a null pointer exception.

apparently the gattDescriptor is most likely not existent.
Checking all the values of the services and characteristics i get seem to be valid, though i do seem to get some empty services (which i skip) when connecting.

As far as I can see in the code, everything should be nicely setup in the library.
Forgive the ugly code as I am first exploring the use of the library, but its not to dissimilar to the original example it is based on. (it is hardwired to find a BLE device called martijn)

Any help would be appreciated as I get the feeling I need to setup something more?

`
using UnityEngine;
using Android.BLE;
using Android.BLE.Commands;
using UnityEngine.Android;
using System.Collections.Generic;
using System.Linq;
using System;

public class CharacteristicInfo
{

    public bool IsSubscribed;
    public string Characteristic;
    public string Service;
}

public class ExampleBleInteractor : MonoBehaviour
{
    [SerializeField]
    private GameObject _deviceButton;
    [SerializeField]
    private Transform _deviceList;

    [SerializeField]
    private int _scanTime = 3;

    private float _scanTimer = 0f;

    private bool _isScanning = false;

    private string m_Martijn = "Martijn";
    private string m_Device = "";
    private bool m_MartijnFound = false;

    private bool m_ConnectCommandSent = false;

    List<CharacteristicInfo> m_Chars= new List<CharacteristicInfo>();

    public void ScanForDevices()
    {
        if (!_isScanning)
        {
            _isScanning = true;
            BleManager.Instance.QueueCommand(new DiscoverDevices(OnDeviceFound, _scanTime * 1000));
        }
    }

    private void Update()
    {
        if(_isScanning)
        {
            Debug.LogWarning("scanning");
            _scanTimer += Time.deltaTime;
[... 120 lines cut ...]
2022/12/06 11:02:31.270 8621 8643 Warn Unity     "command": "DeviceConnected",
2022/12/06 11:02:31.270 8621 8643 Warn Unity     "hasError": false,
2022/12/06 11:02:31.270 8621 8643 Warn Unity     "errorMessage": ""
2022/12/06 11:02:31.270 8621 8643 Warn Unity }
2022/12/06 11:02:31.270 8621 8643 Warn Unity command received for F0:6A:69:EA:13:1C : DeviceConnected
2022/12/06 11:02:31.271 8621 8643 Warn Unity we got device connected
2022/12/06 11:02:31.288 8621 8643 Warn Unity we try to register for data! address 00001800-0000-1000-8000-00805f9b34fb char 00002a00-0000-1000-8000-00805f9b34fb
2022/12/06 11:02:31.288 8621 8643 Warn Unity Queueing Command: SubscribeToCharacteristic
2022/12/06 11:02:31.289 8621 8643 Warn Unity sending command subscribeToCustomGattCharacteristic
2022/12/06 11:02:31.289 8621 8643 Warn Unity Calling Command: subscribeToCustomGattCharacteristic
2022/12/06 11:02:31.306 8621 8639 Debug BluetoothGatt onConnectionUpdated() - Device=F0:6A:69:EA:13:1C interval=36 latency=0 timeout=500 status=0

2022/12/06 11:02:31.363 8621 8643 Error Unity AndroidJavaException: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGattDescriptor.setValue(byte[])' on a null object reference
2022/12/06 11:02:31.363 8621 8643 Error Unity java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGattDescriptor.setValue(byte[])' on a null object reference
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at com.velorexe.unityandroidble.UnityAndroidBLE.subscribeToCustomGattCharacteristic(UnityAndroidBLE.java:384)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at com.unity3d.player.UnityPlayer.nativeRender(Native Method)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at com.unity3d.player.UnityPlayer.access$300(Unknown Source:0)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at com.unity3d.player.UnityPlayer$e$1.handleMessage(Unknown Source:95)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at android.os.Handler.dispatchMessage(Handler.java:105)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at android.os.Looper.loop(Looper.java:227)
2022/12/06 11:02:31.363 8621 8643 Error Unity 	at com.unity3d.player.UnityPlayer$e.run(Unknown Source:20)
2022/12/06 11:02:31.363 8621 8643 Error Unity   at UnityEngine.AndroidJNISafe.CheckException () [0x00096] in <2929d00977a149fa8e08bd559a24d9de>:0 
2022/12/06 11:02:31.363 8621 8643 Error Unity   at UnityEngine.AndroidJNISafe.CallVoidMethod (System.IntPtr obj, System.IntPtr methodID, UnityEngine.jval
2022/12/06 11:02:31.399 8621 8643 Warn Unity we try to register for data! address 00001800-0000-1000-8000-00805f9b34fb char 00002a01-0000-1000-8000-00805f9b34fb
`

WimFaasen, 2022-12-06:
I found the issue.
I was subscribing to characteristics without the descriptor. the ones that did, succeeded.
My apology and consider this closed.

closed 2022-12-06 by WimFaasen

Velorexe, 2022-12-06:
Hi @WimFaasen 

Glad you were able to fix the issue and thank you for using my library!
If you run into any new bugs / errors or want to leave some feedback, then you can always open another issue.