Velorexe/Unity-Android-Bluetooth-Low-Energy #19: license and android source code

adabru, 2023-01-22:
Hi @Velorexe 🙂,

do you have the android source code	in another repository? 
And there is no license? 

Do you know of any other android unity open source project? Someone [asked](https://github.com/adabru/BleWinrtDll/issues/40) in a project I try to support.

Velorexe, 2023-01-22:
Hey @adabru

Yes, the Android source code is also on my profile and can be found right here: [Android Bluetooth Low Energy Java Library](https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy-Java-Library).

Instructions on how to build is there as well, though if you need any help you can always contact me.

For the license, no, there is none. I should probably add one since I've had a previous inquiry about it as well. But it's completely open-source and can be distributed in any product. Credit is optional, appreciated, but not mandatory.

adabru, 2023-01-22:
You were so quick in adding the license 😀 Thanks for the link!

Another question: How did you debug your java code? Did you do it from within Unity?

closed 2023-01-22 by adabru

Velorexe, 2023-01-22:
Unfortunately debugging a Java library from Unity is a mess, and since this is an Android project, debugging in the Editor also sucks. So I had to debug by building every single time, check the logs and check what could have caused the exception. I'm still looking for a better way, so if you find one, please let me know!

adabru, 2023-01-22:
I've never done android development 🤷 What you write sounds frustrating.

You've already mostly separated your unity and android parts I guess. Also the interface is quite completely defined. Maybe it is an option to: 

- Write an extra, small android app to debug your java code from android studio with hot reload.
- Use stubs for the android BLEManager on unity editor side. You could hardcode the Bluetooth responses OR use a Bluetooth dongle and write a little C# wrapper for https://github.com/adabru/BleWinrtDll to match your interface. And contribute your interface to the repository to add android support 😝

Velorexe, 2023-01-22:
Oh actually, now that you mention it, I did use a small android app in the beginning and used that, the only problem with that is that you'd have to remove all Unity related code (getting context and such) from the library before compiling.

I could upload a version that checks if it uses Unity then provides the context in two different ways, then use that along with a lil' app to debug it. I'll look into it in the future, I'm currently looking at if I can turn this project in both a server and client setup.

adabru, 2023-01-22:
> you'd have to remove all Unity related code

Maybe you can provide a unity stub that just doesn't do anything.

Good luck 🙂