Velorexe/Unity-Android-Bluetooth-Low-Energy #26: Handling BLE with multiple unity scenes

auradaej, 2023-04-05:
Hello,

first of all a big thank you for your efforts, just what I needed!

There is only one issue I'm facing: Whenever I change scene in Unity, the BLE stuff doesn't work anymore, and for each command, I get a NullPointerException when calling the library through the BLEManager.

My scenario:

- In Scene 1, I'd like to show some data from BLE device.
- In Scene 2, this is some kind of settings/configuration page, where I connect my BLE device.
- In each scene, I added the BleInteractor and the BleManager as game objects (with the Adapter as the child from BleManager)

In this scenario, the following works just fine:
In Scene 2, I discover and connect to my device, and afterwards I do a READ/SUBSCRIBE.

In this scenario, what does not work is:
In Scene 1, I try to do a READ/SUBSCRIBE command. As a result, I get a NullPointerException when doing Call to the bleLibrary in the Ble Manager. I also checked that none of the parameters are null, they are not.
I already tried:
- Removing these game objects from one of those scenes
- Make BleInteractor a Singleton

Velorexe, 2023-04-05:
Interesting!

I'm not sure if I completely understand your issue, but I have a gist of what's happening. Are you loading these scenes additive to each other? If not, then there's two things you could do:

1) You could try adding a `DontDestroyOnLoad(gameObject)` to both the `BleManager` and `BleAdapter`. That way it stays alive in both scenes whenever you switch between them (though you should only use it Scene 1, so that it transfers over to Scene 2)

2) Create a new scene containing only the `BleManager` and `BleAdapter` and load that one additive, so that it lives alongside Scene 1 and Scene 2 (which I think is the cleaner way).

I hope this solves your issue and if not, we can work something out together!

auradaej, 2023-04-12:
Hi!

Many thanks for the quick response! 

I investigated this and neither of the solutions alone fixed the issue in my case - but both combined works very well now! Thanks!

Velorexe, 2023-04-13:
Haha, interesting behaviour. Wouldn't expect that both together would be the solution, but I'm glad it works.

closed 2023-04-13 by Velorexe