Velorexe/Unity-Android-Bluetooth-Low-Energy #27: Documentation needed

takeraparterer, 2023-07-05:
Hello. I would like to use this to create an app to send and recive data to an arduino with a bluetooth module. the example is able to pair, but how to send and recive data?

Velorexe, 2023-07-05:
Hi @takeraparterer 

Good that you were able to connect. After that you need to find the Characteristic that you want to write, read or subscribe to. If you have both the Service and Characteristic UUID, you can start to read or write data to your device.

Respectively, you'd use:
```cs
var command = new SubscribeToCharacterstic(device, service, characteristic, (byte[] data) => {
   Debug.WriteLine(data.length);
});
```
To subscribe

```cs
var command = new ReadFromCharacteristic(device, service, characteristic, (byte[] data) => {
   Debug.WriteLine(data.length);
});
```
To read

```cs
var command = new WriteToCharacteristic(device, service, characteristic, base64Message);
```
To write.

You'd then queue them up to the `BleManager`, like this:
```cs
BleManager.Instance.QueueCommand(command);
```

Which will execute the command. Do note, that the normal BleCommands take a short-hand UUID, which is the differentiating part in this UUID: `0000[short-hand-uuid]-0000-1000-8000-00805f9b34fb`. If you're using complete custom UUID's, pass a `true` `boolean` to the commands as well.

takeraparterer, 2023-07-06:
Also, how would I go about finding the characteristics?

Velorexe, 2023-07-09:
So within BLE you have a certain amount of actions you can do to get data or write data. These are Subscribing / Notifying, Writing and Reading. For now, these can only be done on Characteristic level. If you download a third party app like LightBlue, you can check what kind of Characteristics are available on the device that you're connecting to.

From there you can check on a Characteristic level what kind of action you can do on it. You'd then take that information and at the point where you've connected to the device, you'd queue up one of the commands that I mentioned in the previous comment.

jidkiiponosBOMJA, 2023-07-18:
Hello. What does "characteristic" in write command mean? And is it necessary to use the abbreviated uuid? How can I shorten it?

Velorexe, 2023-07-18:
Hi @jidkiiponosBOMJA 

The `WriteToCharacteristic` command has two constructors, which only differ in the `data` parameter. The data you send will always be converted to `Base64` and read from the `.jar` file as a `byte array` to the device.

The parameters are explained in the summaries on the constructors themselves, but I'll give a brief rundown:

- `deviceAddress`: The UUID of the device that you used to connect to the device with
- `serviceAddress`: Either the shortened or full length UUID of the Service that you want to write to. Basically the service holds individual Characteristics and is used to label them together.  If you're using a full-length then you'd pass `true` to the `customGatt` parameter.
- `characteristicAddress`: Either the shortened or full length UUID of the Characteristic you want to connect to. If you're using a full-length then you'd pass `true` to the `customGatt` parameter.
- `data`: either a `string` or a `byte[]`. In both cases, it'll be turned in to `Base64` and send to the Android Java library.
- `customGatt`: by default `false`, but indicates if both the `serviceAddress` and `characteristicAddress` are full length UUID's or short hands. `true` for full length, `false` (default) for short.

If your UUID's adhere to the standard `0000[short-hand-uuid]-0000-1000-8000-00805f9b34fb` UUID template for BLE, then you can use the small part in the labeled `[short-hand-uuid]` part for the parameters of `WriteToCharacteristic`.

jidkiiponosBOMJA, 2023-07-18:
Can u give an example of using write command with customGate?

Velorexe, 2023-07-18:
Do you have an overview of the Characteristics available on your HM-10? That makes it easier to write such examples.
For now, I'll pick a generic address.

```cs
// 0x180F -> Battery Service
// 0x2A19 -> Battery Level [under Battery Service]
// Source: https://www.bluetooth.com/specifications/assigned-numbers/

var write = new WriteToCharacteristic(
    _deviceUUID,
// Service UUID
    "0000180f-0000-1000-8000-00805f9b34fb",
// Characteristic UUID
    "00002A19-0000-1000-8000-00805f9b34fb",
// Data
    new byte[] { 0x01 },
// Indicates that these are custom GATT UUID (though the example UUID's could be used with short-hand 16-bit UUID's)
    customGatt: true
);

BleManager.Instance.QueueCommand(write);
```

jidkiiponosBOMJA, 2023-07-18:
Hi,thanks for answer.
I checked my HM-10 UUID:
[image]

In the Internet, I found which uuid corresponds to device name, service name and other. But when I entered these uuids, I got this code:

var write = new WriteToCharacteristic(
               //Device UUID
                "00002A00-0000-1000-8000-00805F9B34FB",
               // Service UUID
               "0000FFE0-0000-1000-8000-00805F9B34FB",
               // Characteristic UUID
               "0000FFE1-0000-1000-8000-00805F9B34FB",
               // Data
               new byte[] { 0x01},
               // Indicates that these are custom GATT UUID (though the example UUID's could be used with short-hand 16-bit UUID's)
               customGatt: false
               //false-long,true-short
           );
            BleManager.Instance.QueueCommand(write);

(This code is'nt working ,it connect with scan button,but dont send commands)

Internet give me this:

[image]

ORANGE-Device UUID
GREEN-Service UUID
YELLOW-Characteristic UUID

Is this correct, or do these UUID belong to others?

Velorexe, 2023-07-18:
Close, but not quite. Your Device isn't a UUID, it's a MAC address. My library will return a the found devices with their MAC address and name if it has one.

Also first the device has to be scanned, so the Android Java library has it cached. Then you have to connect to your device using it's MAC address (done through a button press in the example). 

When the connection is successful, you can queue your commands to the `BleManager`.

Also as you can see for your Service and Characteristic addresses, they follow the same pattern as the standard UUID: `0000[short-hand-UUID]-0000-1000-8000-00805f9b34fb`. So you could use `FFE0` for the Service and `FFE1` for the Characteristic, then leave `customGatt` as false.

jidkiiponosBOMJA, 2023-07-18:
Could you match my data with the Write To Characteristic command? (I'm beginner in unity)

Velorexe, 2023-07-18:
I'm not blaming the fact that I have to write this out on the original issue at hand (the lack of documentation). I recommend picking up some tutorials on C# / .NET and Unity to get familiar with it, but overall, keep practicing!

I'll write out the `WriteToCharacteristic` for you, but I don't know two parameters: the `deviceAddress` and the `data`. I don't know what you're going to send and I don't know the MAC address of your device.

```cs
var command = new WriteToCharacteristic(devieAddress, "FFE0", "FFE1", data);
BleManager.Instance.QueueCommand(command);
```

Again: this is only possible after you scanned for devices and connected to the one you want to send this command to.

Velorexe, 2023-07-24:
@takeraparterer 
Is your original question answered / solved? Else I'm gonna close the issue and you can re-open it if it's not.

takeraparterer, 2023-07-25:
@Velorexe go ahead and close it 👍

closed 2023-07-25 by Velorexe

takeraparterer, 2024-11-17:
Reopening because I'm giving this another shot. I don't understand what I need to do. I'm trying to recreate this python code:
```python
import asyncio
from bleak import BleakClient

address = "D8:B6:73:05:CB:0C"
READ =  "0000FFE1-0000-1000-8000-00805f9b34fb"
WRITE = READ

async def skividi(_,data):
    print(data)

async def main(address):
    async with BleakClient(address) as client:

        await client.start_notify(READ,skividi)

        await asyncio.sleep(1)
        await client.write_gatt_char(WRITE,b"Ping\n")# get battery
        await asyncio.sleep(60)
asyncio.run(main(address))```
but I don't know how