Velorexe/Unity-Android-Bluetooth-Low-Energy #28: Null result for GetInstance method call on librarySingleton

Levieck, 2023-07-14:
Hi, 

_bleLibrary is null after this line of code:
```_bleLibrary = librarySingleton.CallStatic<AndroidJavaObject>("getInstance");```

https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy/blob/main/Assets/Scripts/BLE/BleManager.cs#L160

The project is excecuted in unity editor 2019.4.16f1
I have the following JDK, NDK, SDK settings:
[image]

Do you have an idea why can getInstance return null?

Velorexe, 2023-07-14:
Oof I would have no real clue. Is the `librarySingleton` null as well? Or is it just the `_bleLibrary`? It's possible that the Android package is not compiled with your APK, did you check if it's in the `Android/Plugins` directory?

Levieck, 2023-07-14:
Hi @Velorexe, thanks for the quick answer! You were right, when trying to build and run I got an error message about Java 11 instead of  8 compatibility issue. I'm working on fixing that.