Velorexe/Unity-Android-Bluetooth-Low-Energy #30: Subscribing to multiple characteristics

lucapauri, 2023-08-13:
Hi, I'm trying to use your asset and it works perfectly. The only issue is that I want to subscribe to multiple characteristics but I only get the notifications for the first one. Is there any way I can solve this?
Thanks in advance!

Velorexe, 2023-08-25:
Hi @lucapauri ,

That's an interesting one, I'm not completely sure. I myself haven't tested connecting to multiple Characteristics before, so it might be an oversight that I've missed. If you go into LogCat, does it show that it receives values from the other Characteristics? Or is it limited to just the first one?

From there I can determine if it's a problem with the Java library or if Unity is picking the values up in the wrong way (or just not passing them to the other commands).

ebubar, 2025-01-22:
I have a similar use case and am curious if this was solved? Would also be curious for the script to subscribe to those multiple characteristics that was used.