Velorexe/Unity-Android-Bluetooth-Low-Energy #32: BLUETOOTH_SCAN permission issue (android 11+)

mmar58, 2023-10-07:
Hi, I am running your project on my device and found this issue 

> Exception: AndroidJavaException: java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@4d29c837: GattService registerScanner

And as I was investigating more, I found that on Android 11+ plus devices, this problem is happening because of the permission security change and I also found the [solution](https://stackoverflow.com/questions/67722950/android-12-new-bluetooth-permissions).

But here are some codes which you will have to add inside the jar files. So if you add those and update your project, I will be grateful and it can help a lot of us.

Velorexe, 2023-10-07:
Hi @mmar58 

Thanks for the heads up! I've always had an annoyance against the permissions on different version of Android (and on their API). Just a question, which version of the plugin are you using? There's a [v0.0.3-alpha.0](https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy/releases/tag/v0.0.3-alpha.0) build that's currently experimental, but has overhauled the way permissions are handled. It does require a different way of using the framework, but it should be more straight forward.

If you aren't using the new v0.0.3-alpha.0 version (or can't switch due to restraints), I'll take a look at adding a permissions check and update the `.jar` file accordingly.

mmar58, 2023-10-07:
Hi, I have fixed this issue by asking permission inside Unity. I am creating a pull request now. But hope you can organize the permissions in a good way

closed 2023-10-07 by mmar58

tugsadocal6, 2024-03-18:
> Hi, I have fixed this issue by asking permission inside Unity. I am creating a pull request now. But hope you can organize the permissions in a good way

Hi! How to fixed in Unity? Can you share?