Velorexe/Unity-Android-Bluetooth-Low-Energy #38: Error on write characteristic

lucasmontec-fortis, 2024-02-26:
Im getting this error trying to write a characteristic:
```
2024/02/25 20:57:35.323 12452 12474 Error Unity AndroidJavaException: java.lang.IllegalArgumentException: Invalid UUID string: 000025e1362d-af97-42c1-92e1-41739cc194e6-0000-1000-8000-00805f9b34fb
2024/02/25 20:57:35.323 12452 12474 Error Unity java.lang.IllegalArgumentException: Invalid UUID string: 000025e1362d-af97-42c1-92e1-41739cc194e6-0000-1000-8000-00805f9b34fb
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at java.util.UUID.fromStringJava8(UUID.java:276)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at java.util.UUID.fromString(UUID.java:223)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at com.velorexe.unityandroidble.UnityAndroidBLE.writeToGattCharacteristic(UnityAndroidBLE.java:511)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at com.unity3d.player.UnityPlayer.nativeRender(Native Method)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at com.unity3d.player.UnityPlayer.-$$Nest$mnativeRender(Unknown Source:0)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at com.unity3d.player.UnityPlayer$F$a.handleMessage(Unknown Source:122)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at android.os.Handler.dispatchMessage(Handler.java:102)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at android.os.Looper.loopOnce(Looper.java:230)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at android.os.Looper.loop(Looper.java:319)
2024/02/25 20:57:35.323 12452 12474 Error Unity 	at com.unity3d.player.UnityPlayer$F.run(Unknown Source:24)
2024/02/25 20:57:35.323 12452 12474 Error Unity   at UnityEngine.AndroidJNISafe.CheckException () [0x00096] in <c5ca39a6c98646d4a8b61716838ca615>:0 
2024/02/25 20:57:35.323 12452 12474 Error Unity   at UnityEngine.AndroidJNISafe.
```

My class:
```
public class BLEDeviceController : MonoBehaviour
{
    private const string ServiceUuid = "25e1362d-af97-42c1-92e1-41739cc194e6";
    private const string CharacteristicUuid = "2a7b25d1-cf73-4b3d-bd95-3a44c1646d1c";

//...

    private void SendMovementCommand(string command)
    {
            if (command == _lastCommand && 
                Time.time < _nextRepeatedCommandSend)
            {
                return;
            }
            _lastCommand = command;
            _nextRepeatedCommandSend = Time.time + repeatedCommandThrottling;
            
            var bleCommand = new WriteToCharacteristic(_deviceUuid, ServiceUuid, CharacteristicUuid, command);
            BleManager.Instance.QueueCommand(bleCommand);
    }

}
```

What is that base64 data encoding? Why is that being used?

lucasmontec-fortis, 2024-02-26:
The same error occurs with:
var bleCommand = new WriteToCharacteristic(_deviceUuid, ServiceUuid, CharacteristicUuid, Encoding.UTF8.GetBytes(command));
BleManager.Instance.QueueCommand(bleCommand);

lucasmontec-fortis, 2024-02-26:
Using a custom gatt call:

```
2024/02/25 21:11:18.349 15248 15272 Error Unity AndroidJavaException: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattService android.bluetooth.BluetoothGatt.getService(java.util.UUID)' on a null object reference
2024/02/25 21:11:18.349 15248 15272 Error Unity java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattService android.bluetooth.BluetoothGatt.getService(java.util.UUID)' on a null object reference
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at com.velorexe.unityandroidble.UnityAndroidBLE.writeToCustomGattCharacteristic(UnityAndroidBLE.java:529)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at com.unity3d.player.UnityPlayer.nativeRender(Native Method)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at com.unity3d.player.UnityPlayer.-$$Nest$mnativeRender(Unknown Source:0)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at com.unity3d.player.UnityPlayer$F$a.handleMessage(Unknown Source:122)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at android.os.Handler.dispatchMessage(Handler.java:102)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at android.os.Looper.loopOnce(Looper.java:230)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at android.os.Looper.loop(Looper.java:319)
2024/02/25 21:11:18.349 15248 15272 Error Unity 	at com.unity3d.player.UnityPlayer$F.run(Unknown Source:24)
2024/02/25 21:11:18.349 15248 15272 Error Unity   at UnityEngine.AndroidJNISafe.CheckException () [0x00096] in <c5ca39a6c98646d4a8b6171683
```

Command code:
```csharp
private void SendMovementCommand(string command)
        {
            if (command == _lastCommand && 
                Time.time < _nextRepeatedCommandSend)
            {
                return;
            }
            _lastCommand = command;
            _nextRepeatedCommandSend = Time.time + repeatedCommandThrottling;
            
            var bleCommand = new WriteToCharacteristic(_deviceUuid, ServiceUuid, CharacteristicUuid, Base64Encode(command), true);
            BleManager.Instance.QueueCommand(bleCommand);
        }

        private static string Base64Encode(string plainText) 
        {
            var plainTextBytes = Encoding.UTF8.GetBytes(plainText);
            return System.Convert.ToBase64String(plainTextBytes);
        }
```

Velorexe, 2024-03-04:
@fairdox in the second code block they shared it details that they're using the `customGatt` override.
The Java error message mentions that it can't execute the `getService` method since it's called on a `null` object, which would mean that the `GattServer` (aka the device) is either not known or not connected to the Android device.

farunurisonmez, 2024-05-29:
[image]

I don't understand why the error occurs

lucasmontec, 2024-05-30:
(lucasmontec fortis here, I used the wrong account before)
So, I ended up writing my own library that does the same thing.
It's crude but I was able to do everything.
I still have to write a command queue on the native side, but I have added support for unitask async on unity.
https://gitlab.com/lucasmontec/zuble-unity-ble-library

farunurisonmez, 2024-05-30:
> (lucasmontec fortis here, I used the wrong account before) So, I ended up writing my own library that does the same thing. It's crude but I was able to do everything. I still have to write a command [...]

Have you completed the necessary codes to read the data coming with Serial.print("message")? I want to read the data coming with serial.print() from the HM-10 module through Unity.

lucasmontec, 2024-09-17:
> Have you completed the necessary codes to read the data coming with Serial.print("message")? I want to read the data coming with serial.print() from the HM-10 module through Unity.

Instead of printing the data through serial, just store it and use a characteristic with Read implemented. It's on your client side code to implement how data is stored, accessed and read, not on the BLE implementation.

Velorexe, 2024-12-16:
I'll close this issue for now, as the `customGatt` solution is indeed the way to pass full length UUID's to the library. The new version of this library will improve on the usage of the `customGatt` option by removing it.

closed 2024-12-16 by Velorexe