Velorexe/Unity-Android-Bluetooth-Low-Energy #39: Write characteristic issue with control characters?

fairdox, 2024-03-04:
I'm not sure if this is related to issue #38 because I don't have a Java error. However it seams that the write to characteristic command is not working.
I can connect to the device (an echelon bike) and receive notification but when sending the initialization sequence, the bike does not react. (The same initialization sequence is working fine with a Nodejs setup)

I tried byte array 
`byte[] init3 = { 0xF0, 0xB0, 0x01, 0x01, 0xA2 };`
I tried encoding a String
`string init3 = "\xF0\xB0\x01\x01\xA2";`
or even like this 
`string init3 = "\u00F0\u00B0\u0001\u0001\u00A2";`

I understand that passing byte arrays to Android JNI is not simple, and utf8 encoding is being used instead.
But has anyone succeeded in sending commands that contain control characters and NULLs ?

```
    const string SERVICE_UUID = "0bf669f1-45f2-11e7-9598-0800200c9a66";
    const string COMMAND_UUID = "0bf669f2-45f2-11e7-9598-0800200c9a66";
    _writeToCharacteristic = new WriteToCharacteristic(_deviceUuid, SERVICE_UUID, COMMAND_UUID, init3 , true);
   BleManager.Instance.QueueCommand(_writeToCharacteristic);

```

Velorexe, 2024-03-04:
Heya, thanks for opening an issue.

The library doesn't use UTF8 encoding, but Base64 encoding to send data from and to the Java plugin, so instead, could you try the string `8LABAaI=` and see if that fixes the issue?

I'm thinking of adding a way to send raw `sbyte[]` from Unity to Android using the `AndroidJNI.ToSByteArray` function, which should give the ability to send `sbyte[]` and convert them to Java's `byte[]`, which can then be send to the characteristic.

Howevery, improvements such as these are things I want to add to the "new" v0.0.3-alpha.0 version, instead of improving upon the more legacy project.

fairdox, 2024-03-04:
Wow Base64 encoding did the trick.
thank you so much. !

closed 2024-03-11 by Velorexe