Velorexe/Unity-Android-Bluetooth-Low-Energy #41: No devices appearing on Windows 10

rawatsaurabh, 2024-03-14:
Hello , 
Thanks for sharing this repo.
I just built it using a bit older unity version ( 2019.4.x) , whn I click the scan i get no devices in the GUI.

[image]

Did I miss some thing
Tx

rawatsaurabh, 2024-03-14:
Hello @Velorexe  , 
Thanks, point noted . I will do the same and update.
Regards,

rawatsaurabh, 2024-03-15:
ok built it and ran it on android and now on clicking scan it shows lots of devices , though there is no scroll view i guess. But it doesn't seems to connect to any one of the device.
Also do I have to use Old input system or new one?

Velorexe, 2024-03-15:
The package is not bound to a specific input system. The example is build using the standard one, but doesn't rely on it (just UI events).

If you open the Android Logcat, the BLE package should log any problems that occur when it does an interaction with Android. From there you can check what kind of errors have occured.

rawatsaurabh, 2024-03-21:
Thanks , it worked for Android.

Velorexe, 2024-03-21:
Glad that you got it to work!
If you have any further questions or inquiries you can always open another issue.

closed 2024-03-21 by Velorexe