Velorexe/Unity-Android-Bluetooth-Low-Energy #43: Option to initiate disconnect from device in refactored code?

thecodechemist99, 2024-03-21:
I’m currently using the refactored version of your project and noticed there is no `Disconnect()` method anymore. I also couldn’t find a method for it in the Java code.
Is there currently an option to initiate a disconnect?

Velorexe, 2024-03-21:
Haha, interesting that I've forgot to add one of the most basic things to the project. Sorry about that, I'll add it this afternoon! Shouldn't be a huge change.

Velorexe, 2024-03-21:
I've released a new `Unity Android Bluetooth Low Energy v.0.0.3-alpha.1` on the Releases tab which contains the `Disconnect` method on the `BleDevice` class.

https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy/releases/tag/v0.0.3-alpha.1