Velorexe/Unity-Android-Bluetooth-Low-Energy #44: Updating to newer Unity editor

nadiDS2, 2024-03-23:
I am trying to bring "Unity-Android-Bluetooth-Low-Energy" up to a newer version of Unity (preferably 2022.3.20f1) and the question I have is:
How much is the "unity.jar" in the "Unity-Android-Bluetooth-Low-Energy-Java-Library" repository tied to 2019.4.16f1?

On a related note, I have three versions of the unity editor and none of them have a "unity.jar".  Where would one find it if one needed a newer version?

Velorexe, 2024-09-20:
Hi Nadi, apologies for the delay in a response.

The code isn't completely tied to 2019.4.16f1, as the Android JNI is backwards and forwards compatible (for now). You can use it in later versions and if you encounter any bugs, I can take a look at them.

I'm unsure where the `unity.jar` is located for your versions, I think Google is your best friend here. Though the `.jar` is only needed if you're going to work on the Java side of things (and compile it yourself).

Velorexe, 2024-12-16:
As it is now a Unity package, it can be imported into any Unity version. Closing this for now.