Velorexe/Unity-Android-Bluetooth-Low-Energy #45: No devices found in one project but same scene works in another project

thecodechemist99, 2024-03-27:
Hi,
I might be missing something really obvious here, but so far I couldn’t figure it out.

I have set up an example scene in one unity project, where everything works fine. After I had some issues with another project, I added the same scene to that project for testing. There, I don’t see any BLE related errors in the logs, but for some reason no devices are discovered.
Both projects have the same Unity version, use your latest alpha release, run on the same device (an Oculus Quest), I added the same AndroidManifest-Patch-Script to both (for the BLE permissions). I even checked the project settings if I could find anything differences that might cause this. I really have no more ideas at the moment.

Any suggestions what else to check or what might cause this? I’d be very grateful for everything ^^.

Velorexe, 2024-03-27:
Interesting. In your app settings, are the permissions successfully given? I know that an error would occur if there's no permissions, but it's worth a shot. Other things that might influence things is the auto-loading of the `.jar` file in the `Plugins/Android` directory, this should be `true`.

Else if all fails (and your project isn't under MDA or other company rules), I can take a look at the project itself as well.

thecodechemist99, 2024-03-28:
> In your app settings, are the permissions successfully given? I know that an error would occur if there's no permissions, but it's worth a shot.

You mean on the Quest? Interestingly enough, my apps don’t show up in the permissions menu (or rather anywhere in the app settings) at all.

> Other things that might influence things is the auto-loading of the `.jar` file in the `Plugins/Android` directory, this should be `true`.

`Load on startup` is set to `true` for the `unityandroidble.jar` in both projects.

> Else if all fails (and your project isn't under MDA or other company rules), I can take a look at the project itself as well.

Thanks for the offer! I’d have to check if that’s possible with my supervisor first.

thecodechemist99, 2024-03-28:
@Velorexe I recreated my entire example project in a fresh unity project. Made sure all the same packages are installed, all the same versions, all the same settings. The only difference left I could find were two different version of the `Version Control` package, but no way to update it in either of the two projects. Everything else, as far as I can tell, should now be basically identical ^^. Unfortunately, the issue stays the same. The existing example project works, the new one doesn’t.

Here’s both of the projects exported as a Unity package file, if you’d like to have a look: [examples.zip](https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy/files/14792634/examples.zip).

yiungyiung, 2024-04-11:
Hey i am getting the same error on my logcat, I have tried diff versions of the plugin 0.2 and the alpha 0.3, unity ver -2022.3.9f
[image]
Were you able to fix the issue ?

thecodechemist99, 2024-04-11:
> Were you able to fix the issue ?

Not yet, unfortunately. But I’ve also been sick for the last weeks, so I haven’t been able to work further on this since I opened the issue.
I have one thing still on my mind, though. I remember vaguely that I noticed while testing that the name of the meta SDK changed somewhere within or between my projects. I think the working version originally still had it named Oculus in the project settings and now it’s something with Meta. So maybe some change in the SDK is to blame here? Still have to further investigate this, though.

thecodechemist99, 2024-05-06:
> Seems to be an issue that's on the .NET part

I actually suspect my original issue to be on the Java side. I have no idea what caused the `NullReferenceException` for me, but now after once again updating all packages in all three projects, they all behave the same again and produce the issue I originally described: No errors in the log but also no devices detected.

I’ve noticed I never shared a log of the original behaviour, sorry about that. Here’s what it looks like:

[image]

From the logs, everything seems to work fine on the .NET side. The task is created, sent to the Java library and just nothing ever comes back. The only issue on the .NET side I can imagine is if the Java library sends something back but it isn’t correctly detected.

thecodechemist99, 2024-05-06:
I figured out how to better filter the logs and found some additional BLE-related logs in between:

[image]

MaxPalmer-UH, 2024-05-08:
I and another developer in my team see the same issue - it used to scan for devices fine, now on his machine and his android device we see it start to scan and hen stop the scan with no devices found. No errors in the log output. We've tried rebooting the phone, restarting bluetooth, uninstalling and resinstalling the app (giving permissions again), logging out permission issues (no errors), changing from IL2CCP to NET and back - but we never get back the behaviour where devices are found on a scan. It's really baffling. Is there any low level logging that can be enabled, perhaps it's on the Java side.

@thecodechemist99 did you ever get to the bottom of this. One moment it was working and then never works again...

MaxPalmer-UH, 2024-05-08:
I also now get the same behaviour on my phone, it scans but no devices are ever found.

MaxPalmer-UH, 2024-05-08:
OK, some things that helped.
1. Make sure location services are on.
2. You may need to manually grant fine location permission to your app. There is some suggestion online that in more recent version of Android that this cannot be granted by placing it in your manifest file.

paulhayes, 2024-12-02:
Did anyone get any further with idenfiying the cause of this?

BillyPung, 2025-03-03:
I also met the NullReference of BLEManager when doing the SendTask, under the branch _reflector/project_overhaul_. And I’m using the Unity version 2022.3. Has anyone figured it out?

paulhayes, 2025-03-03:
Hi @BillyPung, Did you call BLEManager.Initialize() first?