Velorexe/Unity-Android-Bluetooth-Low-Energy #46: Unable to re-scan after device is found

gepz, 2024-03-28:
I'm using the latest alpha version and noticed that if I call SearchForDevicesWithFilter and a device is found and I decide to do nothing, the call to the function no longer works.
Below is a simplified version of the code I ran. Could I ask you to do a quick check and note what I did wrong here if you see anything?
```C#
public class DeviceA: MonoBehaviour
{
    public string DeviceAddress;
    Coroutine scanningRoutine = null;

    public void StartScanning()
    {
        scanningRoutine = StartCoroutine(ScanForDevices());
    }

    IEnumerator ScanForDevices()
    {
        int scanDuration = 5;
        Scanning?.Invoke();
        while (true)
        {
            BleManager.Instance.SearchForDevicesWithFilter(
                scanDuration * 1000,
                OnDeviceFound,
                deviceMac: DeviceAddress
            );
            yield return new WaitForSeconds(scanDuration + 0.1f);
        }
    }

    private void OnDeviceFound(BleDevice device)
    {
        StopCoroutine(scanningRoutine);
        scanningRoutine = null;
        StartScanning();
    }
}
```

Velorexe, 2024-04-23:
Heya,

I'll do a quick check as to why the alpha version is behaving this way, haven't encountered it myself yet, but I'll try finding out what is causing it.

nadiDS2, 2024-06-28:
I'm having a similar problem:  if the BLE connection is lost, neither scanning nor re-connecting work.
Very simple function (debug prints, etc stripped):

```
        private void OnDisconnected(BleDevice device)
        {
            // unsubscribe if subscribed
            if (_txSubscribed)
            {
                _txCharacteristic.Unsubscribe();
                _txSubscribed = false;
            }
            // there is no callback so add a short delay
            DelaySec(0.1f);

            // restart
//            BleManager.Instance.SearchForDevicesWithFilter(10 * 1000, OnDeviceFound, OnSearchStop, serviceUuid: NusUUID);
            _bleDevice.Connect(OnConnected, OnDisconnected);
        }
```
Neither of the options above ever results in a callback.

Velorexe, 2024-06-30:
Thanks for the information!
Is there anything in Android LogCat that shows that the device is disconnected? Or perhaps any errors? Producing it locally I didn't get the same result.