Velorexe/Unity-Android-Bluetooth-Low-Energy #49: Can't properly connect or read characteristic from BLE sensor

mmar58, 2024-04-19:
Hi, I am was testing this plugin and tried to connect with a BLE sensor. But it's not properly connecting and not reading the characteristic values. I have tested with your temperature characteristic and then tested with Heart rate service UUID and characteristic code, but same result. I have attached the screenshot from device.[image]

Here is the console log 
[Uploading Error  log.txt…]()

mmar58, 2024-04-19:
[image]

Velorexe, 2024-04-23:
Heya,

Seems like you're trying to connect to a characteristic on a service that isn't on the device (or can't be found). Can you show me the C# code that you're using to read from the characteristic?

farunurisonmez, 2024-05-29:
[image]

I am getting the same error, @Velorexe 

[image]

paulhayes, 2024-09-25:
Hi @mmar58, did you resolve your issue?

mmar58, 2024-09-25:
No, I have paused using this asset and waiting for the fix to resume work
with this asset.

paulhayes, 2024-09-25:
As we can't reproduce you bug, we can't provide a fix.

Can you try the latest version of the project, using Unity Package Manager and "Add package from git url". If the problem persists, can you try creating an example project, that reproduces the problem, and then zip it up and post it here.

mmar58, 2024-09-25:
Hi @paulhayes , thanks for considering about this. I will test the latest project and give you feedback about this and if this problem persist, I will share the demo, so that you can test it too.

SaaskunDevs, 2024-11-20:
Hi! I have the same problem, triying to connect a LED strip.
I just import the repostory and test the sample scene.

The error ocurse before the SubscribeToExampleService function

paulhayes, 2024-11-22:
Hi @SaaskunDevs which version of the project are you using? Are you importing via Package Manager or the old unity package file under releases?

SaaskunDevs, 2024-11-27:
> Hi @SaaskunDevs which version of the project are you using? Are you importing via Package Manager or the old unity package file under releases?

I'm Using the Package Manager

paulhayes, 2024-11-27:
Great

Can you provide Unity Version, android SDK, so I can try and reproduce the issue.