Velorexe/Unity-Android-Bluetooth-Low-Energy #52: Cannot scan or try out the plugin in Unity Version 2022.3.9

yiungyiung, 2024-05-03:
idk if i am missing anything but i cant run the application
[image]

paulhayes, 2024-05-03:
Hi @yiungyiung, what version of the package are you running?

Can you also provide what Android version you are targeting in Unity.

yiungyiung, 2024-05-04:
I had tried it out for 34 and 33 
and minimum was 14

yiungyiung, 2024-05-04:
Hey i manged to connect to it and 
using this to print data

using UnityEngine;
using Android.BLE;
using Android.BLE.Commands;

public class BleDataReceiver : MonoBehaviour
{
    private string deviceAddress = "34:B7:DA:52:98:21";
    private string serviceUUID = "4fafc201-1fb5-459e-8fcc-c5c9c331914b";
    private string characteristicUUID = "beb5483e-36e1-4688-b7f5-ea07361b26a8";

    private void Start()
{
    // Scan for Bluetooth devices
    DiscoverDevices scanCommand = new DiscoverDevices();
    BleManager.Instance.QueueCommand(scanCommand);
    Debug.Log("inst");
    // Connect to the ESP32 device after a short delay
    Invoke("ConnectToDevice", 5f);
}

private void ConnectToDevice()
{
    ConnectToDevice connectCommand = new ConnectToDevice(deviceAddress, OnDeviceConnected);
    BleManager.Instance.QueueCommand(connectCommand);
}

    private void OnDeviceConnected(string deviceAddress)
    {
        // Subscribe to the characteristic
        SubscribeToCharacteristic subscribeCommand = new SubscribeToCharacteristic(deviceAddress, serviceUUID, characteristicUUID, HandleCharacteristicChanged);
        BleManager.Instance.QueueCommand(subscribeCommand);
    }

    private void HandleCharacteristicChanged(byte[] value)
    {
        // Handle the received data from the characteristic
        string receivedData = System.Text.Encoding.UTF8.GetString(value);
        Debug.Log("Received data: " + receivedData);
    }
}

in logcat I am getting this 
 AndroidJavaException: java.lang.IllegalArgumentException: Invalid UUID string: 00004fafc201-1fb5-459e-8fcc-c5c9c331914b-0000-1000-8000-00805f9b34fb
05-04 23:53:53.651 24011 24041 E Unity   : java.lang.IllegalArgumentException: Invalid UUID string: 00004fafc201-1fb5-459e-8fcc-c5c9c331914b-0000-1000-8000-00805f9b34fb
05-04 23:53:53.651 24011 24041 E Unity   :      at java.util.UUID.fromStringJava8(UUID.java:276)
05-04 23:53:53.651 24011 24041 E Unity   :      at java.util.UUID.fromString(UUID.java:223)
05-04 23:53:53.651 24011 24041 E Unity   :      at com.velorexe.unityandroidble.UnityAndroidBLE.subscribeToGattCharacteristic(UnityAndroidBLE.java:322)
05-04 23:53:53.651 24011 24041 E Unity   :      at com.unity3d.player.UnityPlayer.nativeRender(Native Method)
05-04 23:53:53.651 24011 24041 E Unity   :      at com.unity3d.player.UnityPlayer.-$$Nest$mnativeRender(Unknown Source:0)
05-04 23:53:53.651 24011 24041 E Unity   :      at com.unity3d.player.UnityPlayer$C$a.handleMessage(Unknown Source:122)
05-04 23:53:53.651 24011 24041 E Unity   :      at android.os.Handler.dispatchMessage(Handler.java:102)
05-04 23:53:53.651 24011 24041 E Unity   :      at android.os.Looper.loopOnce(Looper.java:230)
05-04 23:53:53.651 24011 24041 E Unity   :      at android.os.Looper.loop(Looper.java:319)
05-04 23:53:53.651 24011 24041 E Unity   :      at com.unity3d.player.UnityPlayer$C.run(Unknown Source:24)
05-04 23:53:53.651 24011 24041 E Unity   :   at UnityEngine.AndroidJNISafe.CheckException () [0x00000] in <00000000000000000000000000000000>:0
05-04 23:53:53.651 24011 24041 E Unity   :   at UnityEngine.AndroidJava

paulhayes, 2024-05-05:
If you are planning on using full length UUID, please set the customGatt parameter to true when calling the various BLE methods. 

The default is expecting the sort 16bit shortcode UUID.

thecodechemist99, 2024-05-06:
@yiungyiung what did you change in your code to be able to connect now?
I’m using the refactored alpha version, so the syntax and plugin structure is different, but maybe being able to pinpoint what fixed it for you still helps me to resolve the issue, too.

bzonca, 2024-08-19:
> Hey i manged to connect to it and using this to print data
>
> using UnityEngine; using Android.BLE; using Android.BLE.Commands;
>
> public class BleDataReceiver : MonoBehaviour { private string [...]

How did you manage to connect to it? I am getting the same error on a fresh install.

paulhayes, 2024-09-25:
To @bzonca and anyone else with the "Invalid UUID string" error. Be aware that when using a full length UUID ( not the 16bit short UUID ) you must set the optional customGATT argument in the command constructor to ```true```.

```new SubscribeToCharacteristic(deviceAddress, serviceUUID, characteristicUUID, HandleCharacteristicChanged, customGatt=true)```

closed 2024-09-25 by paulhayes