Velorexe/Unity-Android-Bluetooth-Low-Energy #53: Devices not listed out when clicked on scan in the android mobile and also on Quest2

VAkshathaPrasad, 2024-05-16:
Hi
I have downloaded the package from releases tab dated March 20 2023. 
I tried to build and run successfully. I am able to upload the .apk file to the android phone which has version 13.
App is running in the mobile but when clicked on scan, no devices are listed out. Screenshot of it is also attached here.
[image]

Unity version - 2022.3.25f1
API level - tried multiple levels - 22,32,33(which is my Android mobile's version)

Any help is appreciated.

Cnaits, 2024-05-27:
Did you get any errors? 
Might be how Bluetooth permissions are (or aren't) set up. I handle permissions in Unity with this:

```
if (!Permission.HasUserAuthorizedPermission(Permission.CoarseLocation) ||
    !Permission.HasUserAuthorizedPermission(Permission.FineLocation) ||
    !Permission.HasUserAuthorizedPermission("android.permission.BLUETOOTH_SCAN") ||
    !Permission.HasUserAuthorizedPermission("android.permission.BLUETOOTH_CONNECT") ||
    !Permission.HasUserAuthorizedPermission("android.permission.BLUETOOTH") ||
    !Permission.HasUserAuthorizedPermission("android.permission.BLUETOOTH_ADMIN"))
{
    Permission.RequestUserPermissions(new string[]
    {
        Permission.CoarseLocation,
        Permission.FineLocation,
        "android.permission.BLUETOOTH_CONNECT",
        "android.permission.BLUETOOTH",
        "android.permission.BLUETOOTH_ADMIN"
    });
}
```

There are some more permissions than currently in the manifest. You might need to add them there too.

paulhayes, 2024-09-20:
Hi @VAkshathaPrasad did you manage to solve your problems? If not can you try updating to the latest package manager compatible version of the library and see if this solves your problem.

closed 2024-09-25 by paulhayes