Velorexe/Unity-Android-Bluetooth-Low-Energy #55: No error provided when UUID using capitals are provided, but they will stop command responses from firing callbacks

paulhayes, 2024-06-26:
Quick description: providing Service or Characteristic UUIDs containing capital letters for Commands such as ReadFromCharacteristic, will result in call backs not firing. 

I appreciate that there's several easy solutions to this. In fact I'm more than happy to do this fix. However as I've still got forked pull requests waiting, I don't want to start piling up pull requests that won't see the light of day.

@Velorexe, this is an awesome project, don't let it die because you don't have the time to maintain it. I'd really love to help take on some of the load maintaining this project. You would still have final say about fixes, and future features. But at least this way we don't get stuck.

Happy to talk more about it

Velorexe, 2024-06-30:
Hi @paulhayes ! Defintely won't let this die, but it is like you said. Time (and motivation) is hard to find for this project, though I do see that a lot of people have a use for this project. I think it's a good idea that we take this private and start looking at how we can bring the `project-overhaul` branch to `main` soon.

wupy36, 2024-08-12:
Hi @paulhayes is this why callback events are updating such as the read or subscribe?

wupy36, 2024-08-12:
@paulhayes what would be an easy fix for a UUID that doe have upper cases? Would this be on the side of adjusting the java plugin or the Unity C# side?

paulhayes, 2024-08-12:
> Thank you Paul, I will see what I can do to make the uuid lower case.
> 
> for an example if the uuid does contain an upper case will it prevent the Debug.Log from firing in this example from devicebutton.cs?
> 
> ` public void SubscribeToExampleService() { //Replace these Characteristics with YOUR device's characteristics _readFromCharacteristic = new ReadFromCharacteristic(_deviceUuid, "1848", "03C4", (byte[] value) => { Debug.Log(Encoding.UTF8.GetString(value)); }); BleManager.Instance.QueueCommand(_readFromCharacteristic); }`
> 
> ![image](https://private-user-images.githubusercontent.com/5973921/357082196-ae13d546-56e2-4319-817d-8281c229f0a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NzU2OTMsIm5iZiI6MTcyMzQ3NTM5MywicGF0aCI6Ii81OTczOTIxLzM1NzA4MjE5Ni1hZTEzZDU0Ni01NmUyLTQzMTktODE3ZC04MjgxYzIyOWYwYTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTJUMTUwOTUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTJmMTM0NGU4NjRhMjY4NDRmZTdjZWQ4Y2VhZDNkZDc5YTRiMjFjM2YxMGY0YjZkMzBmOTljYjVmNDE1ZWY2MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.SNPcvnehQKXnq2cuukU4K5PutSVxR_66yFjkJQKrAsE)

As UUIDs are case insensitve hex strings, simply use string.ToLower() before passing them into the library.

If you want to fix it in the library, in the constructor of any commands such as ```ReadFromCharacteristic```, simply convert the UUID to lower.
```
-            DeviceAddress = deviceAddress;
-            Service = serviceAddress;
-            Characteristic = characteristicAddress;
+            DeviceAddress = deviceAddress.ToLower();
+            Service = serviceAddress.ToLower();
+            Characteristic = characteristicAddress.ToLower();
```

wupy36, 2024-08-12:
@paulhayes I was scratching my head for a bit and then realized that the DeviceAddress is a Mac and does need to be all caps. Only the Service and Characteristic needs to be forced into a ToLower() this made the subscribe events function and the callbacks work.

paulhayes, 2024-08-13:
ooops sorry about that! I was a bit overzealous with the string conversion wasn't I!

closed 2024-09-25 by paulhayes