Velorexe/Unity-Android-Bluetooth-Low-Energy #57: Device information added RSSI

Wade10729, 2024-08-20:
Hello, I am currently looking for how to get RSSI, MAC, device name and other information in the scanning interface. I want to know how to add RSSI and other information to the information scanned in this project.

Velorexe, 2024-09-20:
Hi Wade, apologies for the late response.

The current version of the plugin doesn't support getting RSSI information. However, getting the MAC address and device name is currently supported, as those will be provided in the callback when scanning for new devices.

Wade10729, 2024-10-07:
Thank you for your response. It seems that I need to find other ways to determine the distance of the Bluetooth transmitter. I appreciate your reply.

paulhayes, 2024-12-01:
Hi @Wade10729, RSSI will be included. It's currently a feature in the experimental refactor/project-overhaul branch.

closed 2024-12-01 by paulhayes

Wade10729, 2024-12-09:
Thank you so much, I can proceed with my testing now.