Velorexe/Unity-Android-Bluetooth-Low-Energy #58: Subscribing to Characteristic

rrajagop99, 2024-09-02:
Hello
Need some guidance please -
1. I got the service and character uuid from the wearable device.  But where do I enter them?
2. Also, in Unity, do I get the BLE Manager in the Hierarchy or add as a component to one of the Text Game Object in my scene where I want to see these characteristics in my UI?
thank you.

paulhayes, 2024-09-02:
Hi @rrajagop99 ,

So this library does require writing code.  Have you seen the example scene in Samples/Scenes/BluetoothLowEnergyExample? It gives you an example of scene structure and how to interface with the library in your own code.

rrajagop99, 2024-09-08:
Hello @paulhayes 
Thank you for your response.  I have seen the example scene and have got them into my scene with BLEManager, BLEAdapter etc. in the Unity Hierarchy.  But on clicking on "Scan" button, nothing happens which is why I was asking where do I enter the service and character uuid etc. of the wearable device in Unity? Some more details on writing code in library would be appreciated.
thank you

paulhayes, 2024-09-20:
It sounds like you have a permissions issue. What version of Android are you publishing to, and what version of Unity?

Have you used android logcat package, to spot any errors or warnings being thrown?

paulhayes, 2024-09-20:
Hi @rrajagop99, I've just made some improvements to the code base, can you update the package, and open the BLE Library post install wizard Window->"Android Bluetooth Low Energy Library" and make sure your AndroidManifest.xml entries are correct.

rrajagop99, 2024-09-21:
@paulhayes 
Hello sir, 
I am publishing to Android 12, Unity 2021.3.11f1, exporting from Unity, building in android Studio Dolphin. I have bluetooth, bluetooth-admin, bluetooth-scan, bluetooth-connect, bluetooth-advertise, all the three location permission in Android manifest as well.   It is just not connecting to the wearable.  Logcat does not give any meaningful error that are related to the connectivity or device but I will keep looking. thanks a lot.

paulhayes, 2024-09-22:
Did you try with the latest version of the library, v0.0.5?

When you hit the scan button the first time the app is installed, are you presented with the allow bluetooth app permission prompt? The only permissions you should need are ```<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />``` and ```<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />```

We haven't tested building from Android Studio. It shouldn't make a difference. What is the reason you are building from Android Studio?

Also would you mind posting your logcat here, from when you run the app, to a second after you hit the scan button.