Velorexe/Unity-Android-Bluetooth-Low-Energy #60: Can't read characteristic values

JaviNunsys, 2024-09-10:
I'm having an issue and i don't know how to solve it. I'm discovering, filtering and connecting successfully to my devices, but when I try to get the data, the android logcat don't debug any information. I'm using the same code of the sample but changing the Uuid. I made a test for that characteristic with nrf connect and got what you can see in the second image. Any hint on how to solve this? Is it because of disabled notifications? If so, do you know how can i enable it?

[image]

[image]

paulhayes, 2024-09-20:
Hi @JaviNunsys , we've got a known bug at the moment, can you try using lowercase server and characteristic UUID and see if that works. I'm writing a pull request but I don't have write access to the repo so have to wait for Velorexe to approve.

paulhayes, 2024-12-01:
Did that fix it @JaviNunsys? I'll repopen if you are still having issues.

closed 2024-12-01 by paulhayes

JaviNunsys, 2024-12-02:
Hi! I forgot to post the solution. The problem was the execution order. I wasn't waiting enough for the sensor to be ready before trying to subscribe the characteristic