Velorexe/Unity-Android-Bluetooth-Low-Energy #64: Subscription callback stop receiving new values after writing bytes to the same characteristic

gepz, 2024-10-01:
The callback (OnValue in the below example) stops receiving new values after writing bytes to the characteristic using the `Write` method (OnButtonAClick in the below example).

```
using System.Collections;
using System.Collections.Generic;
using Android.BLE;
using TMPro;
using UnityEngine;
using UnityEngine.UI;

public class DemoDevice : MonoBehaviour
{
    public string DeviceAddress;
    public TextMeshProUGUI InfoText;
    string serviceUuid = "ffe0";
    string charactersticUuid = "ffe1";
    Coroutine scanningRoutine = null;
    BleGattCharacteristic characteristic = null;
    BleGattService service = null;

    void Start()
    {
        StartScanning();
    }

    void Update() { }

    public void StartScanning()
    {
        scanningRoutine = StartCoroutine(ScanForDevices());
    }

    IEnumerator ScanForDevices()
    {
        int scanDuration = 5;
        InfoText.text = "Scanning...";
        while (true)
        {
            BleManager.Instance.SearchForDevicesWithFilter(
                scanDuration * 1000,
                OnDeviceFound,
                deviceMac: DeviceAddress
            );
            yield return new WaitForSeconds(scanDuration + 0.1f);
        }
    }

    private void OnDeviceFound(BleDevice device)
    {
        InfoText.text = "Device Found";
        device.Connect(OnConnected, OnDisconnected);
    }

    private void OnConnected(BleDevice device)
    {
        InfoText.text = "Connected";
        StopCoroutine(scanningRoutine);
        scanningRoutine = null;
        service = device.GetService(serviceUuid);
        characteristic = service.GetCharacteristic(charactersticUuid);
        characteristic.Subscribe(OnValue);
    }

    private void OnDisconnected(BleDevice device)
    {
        InfoText.text = "Disconnected";
        characteristic = null;
        service  = null;
        StartScanning();
    }

    private void OnValue(byte[] value)
    {
        string receivedString = System.Text.Encoding.UTF8.GetString(value);
        InfoText.text = $"Received Value: {receivedString}";
    }

    public void OnButtonAClick()
    {
        if (characteristic != null)
        {
            byte[] bytes = System.Text.Encoding.UTF8.GetBytes("ABCD");
            characteristic.Write(bytes);
        }
    }

}
```

paulhayes, 2024-10-03:
Hi, thanks for reporting this bug. It looks like you are using the new project-overhaul branch. This branch is still being developed, so it's probably not production ready. But reports like this are really helpful for testing, so please keep helping us test.

I'm curious about using a characteristic as both a subscription and a writable. What's the usecase?

gepz, 2024-10-03:
Thanks for the quick response! I can confirm that the bug is no longer present in version 0.0.5. I was initially using 0.0.3-alpha-1 when I encountered the issue, but after updating to the latest 0.0.5 release, everything is working as expected.

Regarding the use case, I'm working on a project to remotely control a sensor using a Bluetooth module. The characteristic serves as both a subscription for receiving sensor data and a writable for sending control commands.

closed 2024-12-01 by paulhayes