Velorexe/Unity-Android-Bluetooth-Low-Energy #65: Connect Quest 3 with an Arduino

7Flo117, 2024-10-13:
Hi 
I would like to create an Unity program which connects my Quest 3 with an Arduino via BLE. Is it possible to do that with this Plugin? 
I tried the Example scene but when I click on the Scan Button the app crashes.

aloHyomora, 2024-11-10:
@7Flo117 hi 7Flo117!
Did you solve the problem well?

paulhayes, 2024-12-01:
Can you use adb logcat or Unity's Logcat package to record the trace log before the crash, and then post it here as a code block.

Also please include Unity Version and selected Android SDK you have selecting in Player settings..

paulhayes, 2024-12-03:
There is a setup wizard that helps you add the correct flags to you AndroidManifest xml file. It should appeared when you installed. You can open it from the menu here: ```Window/Android Bluetooth Low Energy Library```