Velorexe/Unity-Android-Bluetooth-Low-Energy #70: Subscription has a problem recognizing messages.

MichauCorp, 2025-01-22:
Hello, im exepriencing some issues in making the subscription work with my arduino.

I have an arduino sending incremental messages every second: "hello - 1" , "hello - 2" , "hello - 3" , etc...

I wrote a small ConnectionManager script and when i tried using the subscribe to characteristic command, it either didnt work at all, or didnt invoke the "CharacteristicChanged" whenever a message was recieved.

the thing is, the BleManager does recieve and log the messages live, and when i try using the ReadFromCharacteristic command to read "manually", it does work.

I might be missing something basic and if it will help ill share the code, but for now i couldnt get it to work so I wanted to ask here for assistance.

thanks in advance :)

paulhayes, 2025-01-22:
Can you attach your ConnectionManager class. I'd like to take a look.

MichauCorp, 2025-01-22:
here is the current version:

`using System;
using UnityEngine;
using Android.BLE;
using Android.BLE.Commands;
using TMPro;
using System.Text;

public class BLEConnectionManager : MonoBehaviour
{
    public TextMeshProUGUI messageText;
    public bool read = true;
    public bool write = false;
    public bool subscribe = false;

    [Header("BLE Settings")]
    public string targetDeviceName = "BLE messenger"; // Name of your Arduino BLE device
    public string messageServiceUuid = "19B10000-E8F2-537E-4F6C-D104768A1214"; // your service UUID
    public string messageCharacteristicUuid = "19B10001-E8F2-537E-4F6C-D104768A1214"; // your characteristic UUID

    private string targetDeviceAddress = null;
    private bool isConnected = false;
    private DiscoverDevices discoverCommand;
    private ConnectToDevice connectCommand;
    private SubscribeToCharacteristic subscribeCommand;
    private ReadFromCharacteristic readCommand;
    private WriteToCharacteristic writeCommand;

    private void Start()
    {
        // Ensure the BLE Manager is initialized
        if (!BleManager.IsInitialized)
        {
            BleManager.Instance.Initialize();
        }

        // Start scanning for devices
        Debug.Log("Starting BLE device discovery...");
        BleManager.Instance.QueueCommand(discoverCommand = new DiscoverDevices(OnDeviceDiscovered, OnDiscoveryFinished));
    }

    private void OnDeviceDiscovered(string deviceAddress, string deviceName)
    {
        Debug.Log($"Discovered Device: {deviceName} - {deviceAddress}");

        // Check if the discovered device matches the target device name
        if (deviceName == targetDeviceName)
        {
            Debug.Log($"Target device found: {deviceName}");
            targetDeviceAddress = deviceAddress;

            // Stop scanning and connect to the device
            discoverCommand.End();
            ConnectToDevice();
        }
    }

    private void OnDiscoveryFinished()
    {
        Debug.Log("Device discovery finished.");
        if (targetDeviceAddress == null)
        {
            Debug.LogWarning("Target device not found during discovery.");
        }
    }

    private void ConnectToDevice()
    {
        if (targetDeviceAddress == null)
[... 125 lines cut ...]
    {
        if (deviceAddress == targetDeviceAddress)
        {
            Debug.LogWarning($"Disconnected from device: {deviceAddress}");
            isConnected = false;

            // Stop periodic reading
            CancelInvoke(nameof(ReadFromCharacteristic));
        }
    }

    private void OnDestroy()
    {
        // Clean up resources on destruction
        if (isConnected && connectCommand != null)
        {
            Debug.Log("Disconnecting from device...");
            connectCommand.Disconnect(); // Use Disconnect from ConnectToDevice
        }

        // Stop periodic reading
        CancelInvoke(nameof(ReadFromCharacteristic));
    }
}
`

MichauCorp, 2025-01-22:
im also now testing the write command as its written in this version and my arduino doesnt recognize its being written to, even though it works with nrfConnect

jeffgenexr, 2025-04-15:
for me the issue was 
the ids are caps vs non caps

```
                if (CustomGatt)
                {
                    if (string.Equals(obj.Device, DeviceAddress, StringComparison.OrdinalIgnoreCase) &&
                        string.Equals(obj.Service, Service, StringComparison.OrdinalIgnoreCase) &&
                        string.Equals(obj.Characteristic, Characteristic, StringComparison.OrdinalIgnoreCase))
                    {
                        
                        OnCharacteristicChanged?.Invoke(obj.GetByteMessage());
                    }
                }
```