Velorexe/Unity-Android-Bluetooth-Low-Energy #71: Reading Raspberry Pi 4

ebubar, 2025-01-28:
I'm having an issue with getting this to read a BLE signal from a Raspberry Pi 4. I have the correct deviceAddress, serviceUUID and characteristicUUID. I've confirmed it through nrFconnect and bluetoothctl on the pi itself. I know it is broadcasting. Open to any suggestions as I'm hoping its something simple and dumb that i'm just not seeing right now. 

The device shows up in the scan but no connection ever happens.

`using UnityEngine;
using Android.BLE;
using Android.BLE.Commands;
using TMPro;

public class BleDataReceiver : MonoBehaviour
{
    public string deviceAddress = "D8:3A:DD:C3:2B:23";
    public string serviceUUID = "657a6274-0000-e152-c040-7941ae1b73dd";
    public string characteristicUUID = "657a6274-0001-e152-c040-7941ae1b73dd";
    public TextMeshProUGUI statusText;
    public TextMeshProUGUI receivedDataText; // Reference to the TextMeshPro object

    private void Start()
    {
        // Scan for Bluetooth devices
        UpdateStatusText("Starting BLE Scan...");
        DiscoverDevices scanCommand = new DiscoverDevices();
        BleManager.Instance.QueueCommand(scanCommand);
        Debug.Log("inst");
        // Connect to the ESP32 device after a short delay
        Invoke("ConnectToDevice", 5f);
    }

    private void ConnectToDevice()
    {
        UpdateStatusText("Attempting to connect to bcpi");
        ConnectToDevice connectCommand = new ConnectToDevice(deviceAddress, OnDeviceConnected);
        BleManager.Instance.QueueCommand(connectCommand);
        UpdateStatusText("Connected to BCPI?"); 
    }

    private void OnDeviceConnected(string deviceAddress)
    {
        UpdateStatusText("Device Connected");
        // Subscribe to the characteristic
        SubscribeToCharacteristic subscribeCommand = new SubscribeToCharacteristic(deviceAddress, serviceUUID, characteristicUUID, HandleCharacteristicChanged, true);
        BleManager.Instance.QueueCommand(subscribeCommand);
        // Optionally perform a read/write to keep the connection alive
        ReadFromCharacteristic readCommand = new ReadFromCharacteristic(deviceAddress, serviceUUID, characteristicUUID, HandleCharacteristicRead);
        BleManager.Instance.QueueCommand(readCommand);
        UpdateStatusText("Subscribed to service and characteristic");
        Debug.Log("Subscribed to BCPI");
    }

    private void HandleCharacteristicChanged(byte[] value)
    {
        // Handle the received data from the characteristic
        string receivedData = System.Text.Encoding.UTF8.GetString(value);
        Debug.Log("Received data: " + receivedData);
        if (receivedDataText != null)
        {
            receivedDataText.text = "Received Data:" + receivedData;
        }
    }

        private void HandleCharacteristicRead(byte[] value)
    {
        // Convert the byte array to a readable string
        string receivedData = System.Text.Encoding.UTF8.GetString(value);
        Debug.Log("Received data: " + receivedData);

        // Update the UI with the received data
        if (receivedDataText != null)
        {
            receivedDataText.text = "Received Data: " + receivedData;
        }

        // Update the status text to indicate a successful read
        UpdateStatusText("Data read successfully.");
    }
    private void UpdateStatusText(string Text)
    {
        if(statusText != null)
        {
            statusText.text = Text;
        }
    }
}`

Velorexe, 2025-02-03:
Good morning,

From what I'm seeing on the surface nothing is going wrong and you're using a proper setup to connect to the device (i.e. wait until device is found, wait until device is connected and only then proceeding with suscribing).

Generally everything that gets handled inside the library gets logged to LogCat, which will show if something went wrong.
If you could comment part of your LogCat that's relevant to connecting (it'll log which command it's currently executing), then that helps in fixing your issue!

paulhayes, 2025-03-03:
Did you make any progress @ebubar ?

ebubar, 2025-03-11:
I did not. I had to meet a deadline and used a different (pay) plugin. Would like to figure this out but deadlines :/