Velorexe/Unity-Android-Bluetooth-Low-Energy #72: DiscoverDevices not finding devices or finishing after the discover time.

Darpaler, 2025-02-14:
I might be missing something simple. I had it working the other day where I was using a simple script copied from the samples to print out available devices.

`using UnityEngine;
using Android.BLE;
using Android.BLE.Commands;
using UnityEngine.Android;
using NaughtyAttributes;

public class TestBLE : MonoBehaviour
{
    [SerializeField]
    private int _scanTime = 10;

    private float _scanTimer = 0f;

    private bool _isScanning = false;

    [Button]
    public void ScanForDevices()
    {
        Debug.Log("Scanning.");
        if (!_isScanning)
        {
            _isScanning = true;
            BleManager.Instance.QueueCommand(new DiscoverDevices(OnDeviceFound, OnFinishedDiscovering, _scanTime * 1000));
        }
    }

    private void Update()
    {
        if (_isScanning)
        {
            _scanTimer += Time.deltaTime;
            if (_scanTimer > _scanTime)
            {
                _scanTimer = 0f;
                _isScanning = false;
                Debug.Log("Scanning Complete.");
            }
        }
    }

    private void OnDeviceFound(string name, string device)
    {
        Debug.Log($"Device Found: {name}, {device}");
    }
    
    private void OnFinishedDiscovering()
    {
        Debug.Log($"Finished Discovering");
    }
}
`

Now it's not finding any devices, and the OnFinishedDiscovering doesn't get called until I exit play mode, even after the scanTimer runs out.

[image]

Let me know if there's a setting somewhere that might cause this or what other things I can check when you get the chance. Thank you.

Darpaler, 2025-02-14:
I just got a chance to build to the device now. It seems to be working on the device fine, it's just when running it in editor. Does this work in editor and I'm just missing some setting somewhere, or does this only run when built out as an APK?

paulhayes, 2025-02-14:
Hi @Darpaler , your correct this library only connects via the android Bluetooth library, so functions only in builds.

We are considering a bridge application that will pass messages from Unity to an android handset via wifi, which will allow you to test in editor. This is a long term plan however and it may be quite the wait before we implement that feature.

Darpaler, 2025-02-15:
All good. Thank you.

closed 2025-02-15 by Darpaler