Velorexe/Unity-Android-Bluetooth-Low-Energy #73: DiscoverDevices not finding specific device.

Darpaler, 2025-02-18:
I have an Arduino device with a ESP32. I can see it in the bluetooth device list on both my computer and the Meta Quest 3 that I'm building to. When I do DiscoverDevices to try and find it, it doesn't appear. I tried running both before and after pairing the device. I also tried hard coding in the device address for ConnectToDevice, but that doesn't end up connecting either.

I wanted to know if there's a way to just scan for already paired devices. I also wanted to ask how I would go about figuring out why the device will appear in the bluetooth device list, but not in the DiscoverDevices command. Thank you again!

paulhayes, 2025-02-18:
Can you attach your esp32 code to this report for us to take a look at? I
have some esp32 models that I could try this with.

One important thing to note is that the esp device must be advertising, and
then you should not pair the device before running the app.

Finally which device are you running your unity app on, the meta quest? If
so check your logs using ADB  logcat or the unity editor logcat package.
See if you are getting permission errors.

Darpaler, 2025-02-18:
This is what I have in the device setup:
`
String deviceName = "MyDevice";

#include "BluetoothSerial.h"

void setup() {
  Serial.begin(115200);
  SerialBT.begin(deviceName);  // Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}`

When I run:
`BleManager.Instance.QueueCommand(new DiscoverDevices(OnDeviceFound, onFinishedDiscovering, scanTime * 1000));`

I get a list of devices, but "MyDevice" doesn't appear.

[image]

[image]

When I run this:

`_connectCommand = new ConnectToDevice(_deviceUuid, OnConnected, OnDisconnected);
BleManager.Instance.QueueCommand(_connectCommand);`

with MyDevice's address hard coded in, I get this:

[image]

I'm very new Arduino and using Bluetooth in Unity. Preferably I'd like to have the device pre-paired before hand, and then Unity would just check to see if that device is already connected before sending data to it. Thank you!

paulhayes, 2025-02-18:
I see the problem. 

This library is for using Bluetooth Low Energy ( BLE ), but the library you are using is standard Bluetooth BT. These are not cross compatible.

I wouldn't use BluetoothSerial.h unless you have to, it's very power hungry, and is more of a legacy protocol.

What are you trying to achieve with you project? Perhaps I can give some advice.

Darpaler, 2025-02-19:
We're trying to build an output device that just let's you send data from Unity to the device. Things like updating rumble/motors depending on what's happening in the game. Switching off of BluetoothSerial certainly sounds like the right move. Do you know how I'd go about switching off of it/what I should start researching into to get started? Thank you again, I really appreciate the advice.

Darpaler, 2025-02-19:
Honestly, I think the best move is using Unity's new InputSystem and making a custom InputDevice/device layout. I'm just not 100% on that since I'm really only concerned about sending output to the device and not receiving input from it. I was also just having trouble figuring out setting up all of the HID info correctly and getting Unity to recognize it.

paulhayes, 2025-02-19:
Hi Darpaler,

Okay I understand your problem a bit better now. If all you wanted was to make an input device, I'd suggest simply using implementing a generic BLE HID device, like this:

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo/README.md

However asyou want to focus on output, this library is the right one for you. However you have to read up on how BLE works. Like I2C it's an addressed protocol. You define characteristics on the device which are addresses that can be readable writable and notifiable. You define the size and type of data at that address, it can be a single byte, or a float or whatever. 

Characteristics are cluster together as services, this isn't very helpful for most projects, but it's just how the protocol works. I suggest you implement a single service.

There are multiple BLE libraries for esp32. Espressifi have made there own, but there's also implementations by Adafruit and Arduino. So that depends on your framework.

Darpaler, 2025-02-19:
Thank you again for all the help and pointing me in the right direction. One last question for now. How would I go about finding the names and UUIDs of already paired devices. Currently I'm using device name to distinguish the device from other BLE devices. However, in the situation that there are multiple of my device available, I need it to only check for devices that are paired so that it doesn't get confused with other devices with similar names. Thanks.

paulhayes, 2025-02-19:
So unfortunatly we don't have accessed to already paired devices. We are using BLEScanner, however paired devices are communicarted with using the CompanionDeviceManager api. The different is explained here:
[https://developer.android.com/develop/connectivity/bluetooth/ble/background#find-device](
https://developer.android.com/develop/connectivity/bluetooth/ble/background#find-device)

What I tend to do in my projects is the following:

- Put the esp32 device into advertising mode ( so it's detectable by new devices )
- In the android app, use this library to scan for devices
then either of these two options:
- list the device names to the user and let them select which device to connect to
- or automatically connect using the name
then
- store the MAC of the device for future connections
- next time the app is opened run a scan, but this time automatically connect to the device with the matching MAC address 

Also one extra note
We are improving the libray and the refactor branch does allow you to find a device by service UUID. Because you create your own
service UUID, it's a better option for identiftying your device.

closed 2025-03-01 by Darpaler